KipK/Ghettostation #21: arduino1.6.3 unable to compile

lv333ming, 2015-05-26:
GhettoStation.ino:82:33: fatal error: LiquidCrystal_I2C.h: No such file or directory
compilation terminated.

Error message alert

Maarten112, 2015-05-27:
You don't have the right arduino libraries installed to compile the software. You should use the libraries included with ghettostation. Remove all other libraries that refer to liquidcrystal_i2c.h

sniperxnl, 2015-12-16:
I have this problem:
C:\Program Files (x86)\Arduino\libraries\FastSerial/FastSerial.h:47:3: error: #error Must include FastSerial.h before the Arduino serial driver is defined.

 # error Must include FastSerial.h before the Arduino serial driver is defined.

   ^

In file included from C:\Users\Bas\Documents\Ghettostation-master\Ghettostation-master\GhettoProxy\GhettoProxy.ino:17:0:

C:\Program Files (x86)\Arduino\libraries\FastSerial/FastSerial.h:95:25: error: conflicting declaration 'FastSerial Serial'

 extern class FastSerial Serial;

                         ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:224:0,

                 from sketch\GhettoProxy.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:143:25: error: 'Serial' has a previous declaration as 'HardwareSerial Serial'

   extern HardwareSerial Serial;

                         ^

In file included from C:\Users\Bas\Documents\Ghettostation-master\Ghettostation-master\GhettoProxy\GhettoProxy.ino:17:0:

C:\Program Files (x86)\Arduino\libraries\FastSerial/FastSerial.h:306:18: error: conflicting declaration 'FastSerial Serial'

  FastSerial _name(_num,                                          \

                  ^

C:\Program Files (x86)\Arduino\libraries\FastSerial/FastSerial.h:329:40: note: in expansion of macro 'FastSerialPort'

 #define FastSerialPort0(_portName)     FastSerialPort(_portName, 0)

                                        ^

sketch\GhettoStation.h:2:1: note: in expansion of macro 'FastSerialPort0'

 FastSerialPort0(Serial);

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:224:0,

                 from sketch\GhettoProxy.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:143:25: error: 'Serial' has a previous declaration as 'HardwareSerial Serial'

   extern HardwareSerial Serial;

                         ^

exit status 1
Fout bij compileren.

sniperxnl, 2015-12-16:
I install the old Arduino 1.05 R2 and problem is solved

Landin81, 2016-02-07:
I downloaded and installed the 1.05 R2 and its the same problem =/