KipK/Ghettostation #24: No Telemetry

fredericlaurent, 2015-12-21:
managed to compile on mega 2560.... menu, servos,  buttons, and test work... but always "No telemetry" ... my wiring is good... i can use telemetry with Ezio's EZGUI....
have an idea?

fredericlaurent, 2015-12-21:
Ps:  bauds are setted at 57600 and protocol is Multiwii...

Landin81, 2016-02-08:
I have the same problem. I have try direct link between arduino nano TX to arduino MEGA RX.
Ublox selected in both mega and nano, baudrate 57600. Connected a small speaker on the D9 on Nano, can hear the modem signals, same signals on RX pin on MEGA but alot quieter.

"no telemetry" =/
There is no change in the sound if having gps connected to the nano or not, having the gps connected makes the red led flash on the board. No leds are flashing on the mega, is it suppose to?

KipK, 2016-02-08:
OK don't get what you re trying to achieve by putting a speaker on D9. To
use audio telemetry you need an FSK modem module. What's you're listening
on D9 is just softserial data going out. They"re not modulated. You need
the TCM to modulate the signal then.

Maarten112, 2016-02-08:
This isn't a Issue....it's simply the users doing things wrong. GhettoStation works GREAT! Didn't find any issues or bugs myself. Not to offend anyone but almost all of the "issues" posted here are related to users not knowing what they are doing. Please guys, check the wiki/docs, it isn't so hard and everything is documented really well. 

Thumbs up for KipK who have written some awesome code and decided to share it with the world!

Landin81, 2016-02-08:
Well, if the wiki had more info for us that dont know alot of this kind of stuff, it would be easier.
I thought that the low budrate LTM protocol could be sent over audio. There is so little information about this tracker online. Maby a "setup for dummies" is needed. 

So, I need two more arduinos and make TX/RX FSK modems and connect them like: 
Arduino nano (airside) Pin D9 to arduino nano FSK modem, and out to sound on VTX?

MEGA (Groundside). from vtx Audio to arduino nano FSK modem to MEGA RX?

And this will work?

fredericlaurent, 2016-02-09:
Hi Maarten,

Do you use ghetto station with multiwii 2.4 ?

I always get "no telemetry" even when connecting tx rx straight from my 
Flight controler ( crius aio pro )... I must add that my groud station 
works fine with the same tx rx wires...

So where is the problem?

I set protocol to multiwii and baud rate to 57600 just like  my ground 
station.... I don't use modem but APC220....

I agree with you: thumbs up for KipK...

just waiting for your help..

CU

Frederic

fredericlaurent, 2016-02-09:
Hi Guillaume,

glad to hear from you...

as you see, I try to use ghetto station with multiwii 2.4.... and always 
get a "no telemetry" message ... with the same tx rx wiring as my ground 
station ( EZ Gui ) that works fine....
set protocol to multiwii... baudrate to 57600... just like my ground 
station....

I also get "no telemetry" when wiring straight from the FC (crius aio 
pro ) ????

can you please help me ?

cordialement

Frederic

fredericlaurent, 2016-02-09:
I hope Guillaume S will help us....

sniperxnl, 2016-02-13:
I got the same problem.

i buy the FPV modems

And connect it like this:
Aircraft: Gps TX/RX to Ghetto proxy D9 and GND to TX/GND from FPV modem A-out to audio VTX

Ground: Audio goes to Fpv modem IN and RX goes to Arduino Mega 19. Is this the right way?

fredericlaurent, 2016-02-14:
where did you buy the fpv modems?... perhaps I could try this way too...

nelbernal, 2016-02-29:
I always get "no telemetry"  whit any protocol, after downgrade my version of arduino ide to 1.0.6 and compile, Now all work fine.

fredericlaurent, 2016-04-10:
COMPILE WITH older ARDUINO 1.0.6   .... AND IT WILL WORK !!!!!!  THANKS TO
 NELBERNAL  for his help   ...... and  to
GUILLAUME S KIPK for his ..... soft

closed 2016-04-10 by fredericlaurent