KipK/Ghettostation #25: Error

sniperxnl, 2015-12-22:
In file included from GhettoStation.ino:38:
GhettoStation.h:12: error: no matching function for call to 'HardwareSerial::HardwareSerial(int)'
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/HardwareSerial.h:54: note: candidates are: HardwareSerial::HardwareSerial(ring_buffer*, ring_buffer*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/HardwareSerial.h:33: note:                 HardwareSerial::HardwareSerial(const HardwareSerial&)
GhettoStation.h:13: error: no matching function for call to 'HardwareSerial::HardwareSerial()'
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/HardwareSerial.h:54: note: candidates are: HardwareSerial::HardwareSerial(ring_buffer*, ring_buffer*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/HardwareSerial.h:33: note:                 HardwareSerial::HardwareSerial(const HardwareSerial&)

LeoTosti, 2016-01-20:
Hello, I've got the same HardwareSerial error, have you solved this error?
Someone has a solution? 
Thank you.

Landin81, 2016-02-07:
Same problem here. have try several different arduino versions but same problem. =/ spend about 10 hours trying to fix it... oh, sorry, 11 hours now. Thought this would be a easy diy project =(

LeoTosti, 2016-02-07:
Hello I think I have solved the problem, here's the change:

#ifdef TEENSYPLUS2
// This line defines a "Uart" object to access the serial port
HardwareSerial SerialPort1(Serial1); _instead of : HardwareSerial SerialPort1 = HardwareSerial();_
HardwareSerial SerialDebug(Serial); _instead of: HardwareSerial SerialDebug = HardwareSerial();_

After that you should not have error message, but I do not think the code work when using a Teensy. No more error for me with Arduino Mega it compiles ok but I have not yet tested the station because I expected the necessary equipment, being delivered ..

Test with this change and said me if compromait or not the operation of the Ghettostation.
Well, I expect your return. :)