KipK/Ghettostation #29: NO TELEMETRY

Amrello, 2017-04-06:
My problem is that the antenna tracker can not read the protocol ( Error: NO TELEMETRY).
I used different protocols, such as mavlink (multiwii AIO), ublox (NEO6M), different speeds but it did not help. Also, I revised your sketch and turned off the check of received packets but this also did not help. And I understood most of the sketch, but I'm not strong at programming. Maybe you can help me.

Amrello, 2017-04-06:
https://vk.com/id297797756?z=albums297797756 photos

nelbernal, 2017-04-07:
Amrello, check the baudrates, or compile ussing arduino old version.

Amrello, 2017-04-08:
I tried it but it does not work.

Amrello, 2017-04-08:
I'm using 57600 speed

0Balazs, 2017-04-10:
I also have this problem. I tried different protocol, different baud, ghettoproxy. Proxy data is processed and sent to 2400 baud.

inis1981, 2017-08-23:
Can you both post your setups? and method of testing. I've gotten this working, took some time but did get it to work pretty well.

0Balazs, 2017-08-24:
I want to use arduino mega panel. Here is the serial port problem, I had to write the serial port1 to serial port and manually set the speed and the protocol.
If I remember correctly, I could only work that way. But unfortunately I was not ready yet.
Is there any idea why it does not work without poking the program?

Amrello, 2017-08-24:
0Balazs, I also tried to run antenna tracker code but it did not decode mavlink messages Later I started working on my own code for Arduino, and I successfully decode mavlink and write a simple tracking algorithm. If you won't, I can sand working mavlink library and code

piotrone, 2017-09-18:
Hello, I have the same problem. I'm trying to run antenna tracker and meet the error "no telemetry waiting for data...".  Please help, @Amrello, could you share working mavlink library and code ?

inis1981, 2017-09-18:
> piotrone,try this libreries, if it will not be working I can help you to
> write code for your station.
> And try 1.0.6 arduino ide https://www.arduino.cc/en/
> Main/OldSoftwareReleases#previous
> some libreries.zip

Also when testing make sure to hook fc direct to tracker. If it doesn't
work that was something is seriously wrong.

piotrone, 2017-09-18:
> some libreries.zip [1]

hello, thank you very much. I will try and will let you know :) 

 

Links:
------
[1]
https://github.com/KipK/Ghettostation/files/1311577/some.libreries.zip
[2]
https://github.com/KipK/Ghettostation/issues/29#issuecomment-330292899
[3]
https://github.com/

piotrone, 2017-09-18:
hello, thank you very much. I will try and will let you know :) 

 

Links:
------
[1]
https://github.com/KipK/Ghettostation/issues/29#issuecomment-330308190
[2]
https://github.com/

webnamex, 2017-10-25:
@Amrello, hi ,  could you sent the mavlink code to my email(webnamex@gmail.com) ? i want to study it ,thank u

13055ecam, 2019-03-14:
hello, same error 'No telemetry" how can I fix it ?

datamann, 2019-09-28:
For those who are interested, my fork works.
https://github.com/datamann/Ghettostation