KipK/Ghettostation #39: Adding a esp32dev board

knputyi, 2020-07-02:


knputyi, 2020-07-02:
Could someone extend the code for esp32dev board?

Maarten112, 2020-07-02:
You can!

knputyi, 2020-07-02:
> […](#)

What can I do?? 
If I could, I've not been asking...

datamann, 2020-07-02:
@knputyi The way I see it adding support for ESP32 Dev boards will only increase the complexity, the only win will be the formfactor.

knputyi, 2020-07-02:
With mega board and mavlink through Open.HD at 57600 baud rate, the tracker seems to lag from time to time. Isn't  the mega overloaded?

datamann, 2020-07-03:
@knputyi I wouldn't know, if that is the case then you will benefit from adding ESP32 Dev board. The code is quite old so you might need to rewrite some of the code, I know I needed to to get my tracker to work....

whoim2, 2020-10-15:
> С мегаплатой и mavlink через Open.HD на скорости 57600 бод трекер, кажется, время от времени тормозит. Мега не перегружена?

Это мертвый проект, я не смог заставить его нормально работать. Жаль, идея обойтись без компаса и слипринга - хорошая. Может, удастся кому то оптимизировать код..